SzczepanLeon/esphome-components #243: Bad std

tomgst, 2025-07-19:
File : 

driver_apator162.cpp

Line: 

map<string, pair<int, DVEntry>> vendor_values;

Correct value?

std::map<std::string, std::pair<int, DVEntry>> vendor_values;

Can you modify ;)?

SzczepanLeon, 2025-07-23:
On main branch exist correct one.

closed 2025-07-23 by SzczepanLeon

tomgst, 2025-07-23:
Problem w tym, ze uzywalem do tej pory dotychczasowego Twojego repo i wszystko dzialalo do niedawna dla 2 urzadzen. Teraz jak mam 2 urzadzenia apator to czyta dane tylko z pierwszego a drugiego już nie. Sygnal z urzadzenia pobierany jest poprawnie ID itd sie zgadza.

tomgst, 2025-07-23:
`esphome:
  name: wth
  friendly_name: WTH

esp8266:
  board: esp01_1m

# Enable logging
logger:
  level: DEBUG
  logs:
    wmbus: DEBUG
# Enable Home Assistant API
api:
  encryption:
    key: "vXXXXXXXXXXXXXXXX="

ota:
  - platform: esphome
    password: "XXXXXXXXXXXXXXXXXXXXXXXXXX"

wifi:
  ssid: "INEA-8922_2.4G"
  password: "XXXXXXXXXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Wth Fallback Hotspot"
    password: "XXXXXXXXXX"

captive_portal:

time:
  - platform: sntp
    id: time_sntp

external_components:
  - source: github://SzczepanLeon/esphome-components@version_3
    components: [ wmbus ]

wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin:  GPIO14
  cs_pin:   GPIO2
  gdo0_pin: GPIO5
  gdo2_pin: GPIO4
sensor:
#0x00539144 Dziala poprawnie
  - platform: wmbus
    meter_id: 0x00539144
    type: apator162
    key: "00000000000000000000000000000000"
    lqi:
      name: "My lqi"
    rssi:
      name: "My RSSI"
    total_water_m3:
      name: "Woda glowny"
      filters:
        - offset: -82.829    
#Nie pobiera danych : 0x04441453
  - platform: wmbus
    meter_id: 0x04441453
    type: apator162
    key: "00000000000000000000000000000000"
    lqi:
      name: "LQI ogrod"
    rssi:
      name: "RSSI ogrod"
    total_water_m3:
      name: "Woda ogrod"
      filters:
        - offset: +12.000 `

SzczepanLeon, 2025-07-23:
Zmiana sprzętu na nowszy.

A jakie to feature potrzebujesz z nowego ESPHome że downgrade nie wchodzi w rachubę?