SzczepanLeon/esphome-components #249: Error compiling version 4 on ESP32-S3

ScullyMulder38, 2025-08-17:
I wanted to flash an ESP32-S3 with WMBus version 4, but ESPHome stops compiling with the following error:

`INFO ESPHome 2025.7.5
INFO Reading configuration /config/esphome/deltamess-wmz-s3.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing deltamess-wmz-s3 (board: esp32-s3-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.10.2 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_aarch64'
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_aarch64'
Dependency Graph
|-- noise-c @ 0.1.10
|-- wmbus-drivers @ 0.0.0+20250817105618.sha.c856f7d
Compiling .pioenvs/deltamess-wmz-s3/src/esphome/components/wmbus/rf_cc1101.cpp.o
Compiling .pioenvs/deltamess-wmz-s3/src/esphome/components/wmbus/wmbus.cpp.o
Compiling .pioenvs/deltamess-wmz-s3/src/esphome/core/application.cpp.o
Compiling .pioenvs/deltamess-wmz-s3/src/esphome/core/color.cpp.o
In file included from src/esphome/components/wmbus/rf_cc1101.h:9,
                 from src/esphome/components/wmbus/rf_cc1101.cpp:1:
src/esphome/components/wmbus/cc1101_rf_settings.h:3:10: fatal error: ELECHOUSE_CC1101_SRC_DRV.h: No such file or directory

****************************************************************************************
* Looking for ELECHOUSE_CC1101_SRC_DRV.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ELECHOUSE_CC1101_SRC_DRV.h"
* Web  > https://registry.platformio.org/search?q=header:%1B%5Bm%1B%5BKELECHOUSE_CC1101_SRC_DRV.h
*
****************************************************************************************

    3 | #include <ELECHOUSE_CC1101_SRC_DRV.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/deltamess-wmz-s3/src/esphome/components/wmbus/rf_cc1101.cpp.o] Error 1
In file included from src/esphome/components/wmbus/wmbus.cpp:1:
src/esphome/components/wmbus/wmbus.h:17:10: fatal error: WiFiClient.h: No such file or directory

**************************************************************************
* Looking for WiFiClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClient.h"
* Web  > https://registry.platformio.org/search?q=header:%1B%5Bm%1B%5BKWiFiClient.h
*
**************************************************************************

   17 | #include <WiFiClient.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/deltamess-wmz-s3/src/esphome/components/wmbus/wmbus.cpp.o] Error 1
========================== [FAILED] Took 8.38 seconds ==========================`

This error also occurs in versions 3 and 2. What am I doing wrong?
Here is my yaml file:

`esphome:
  name: deltamess-wmz-s3
  friendly_name: DeltaMess_WMZ_S3

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  - platform: esphome
    password: "8702c8ad4f7c934bfb02bd6f841f17d4"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Deltamess-Wmz-S3"
    password: "XXX"

captive_portal:

time:
  - platform: sntp
    id: time_sntp

external_components:
  - source: github://SzczepanLeon/esphome-components@version_4

wmbus:
  all_drivers: True
  log_all: True`

The error listed above occurs at the latest when entering “WM Bus:...”.

SzczepanLeon, 2025-08-17:
Known issue.
Please use v5 or v4 with ESPHome 2025.6.x

closed 2025-08-17 by SzczepanLeon

ScullyMulder38, 2025-08-17:
No chance—even with 2025.6.0 and 2025.6.3—same error in versions 3 and 4:
`esphome run WMZ-Deltamess-S3.yaml
INFO ESPHome 2025.6.0
INFO Reading configuration WMZ-Deltamess-S3.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Core config, version or integrations changed, cleaning build files...
INFO Deleting /home/mike/esphome/.esphome/build/wmz-deltamess-s3/.pioenvs
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.8.2
INFO Compiling app...
Processing wmz-deltamess-s3 (board: esp32-s3-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/LSatan/SmartRC-CC1101-Driver-Lib#master
INFO Installing git+https://github.com/LSatan/SmartRC-CC1101-Driver-Lib#master
git version 2.34.1
Klone nach '/home/mike/.platformio/.cache/tmp/pkg-installing-670fedw8' …
remote: Enumerating objects: 102, done.
remote: Counting objects: 100% (102/102), done.
remote: Compressing objects: 100% (76/76), done.
remote: Total 102 (delta 26), reused 67 (delta 17), pack-reused 0 (from 0)
Empfange Objekte: 100% (102/102), 1.08 MiB | 10.71 MiB/s, fertig.
Löse Unterschiede auf: 100% (26/26), fertig.
Library Manager: SmartRC-CC1101-Driver-Lib@2.5.7+sha.b8c6af4 has been installed!
INFO SmartRC-CC1101-Driver-Lib@2.5.7+sha.b8c6af4 has been installed!
Library Manager: Installing git+https://github.com/SzczepanLeon/wmbus-drivers#1.3.13
INFO Installing git+https://github.com/SzczepanLeon/wmbus-drivers#1.3.13
git version 2.34.1
Klone nach '/home/mike/.platformio/.cache/tmp/pkg-installing-va5mv1xj' …
remote: Enumerating objects: 35, done.
remote: Counting objects: 100% (35/35), done.
remote: Compressing objects: 100% (24/24), done.
remote: Total 35 (delta 25), reused 16 (delta 11), pack-reused 0 (from 0)
Empfange Objekte: 100% (35/35), 15.02 KiB | 1.37 MiB/s, fertig.
Löse Unterschiede auf: 100% (25/25), fertig.
Hinweis: Wechsle zu 'c856f7d1eadb337ae3e6f59f3eb3ca652e371735'.

Sie befinden sich im Zustand eines 'losgelösten HEAD'. Sie können sich
umschauen, experimentelle Änderungen vornehmen und diese committen, und
Sie können alle möglichen Commits, die Sie in diesem Zustand machen,
ohne Auswirkungen auf irgendeinen Branch verwerfen, indem Sie zu einem
anderen Branch wechseln.

Wenn Sie einen neuen Branch erstellen möchten, um Ihre erstellten Commits
zu behalten, können Sie das (jetzt oder später) durch Nutzung von
'switch' mit der Option -c tun. Beispiel:

  git switch -c <neuer-Branchname>

Oder um diese Operation rückgängig zu machen:
  git switch -

Sie können diesen Hinweis ausschalten, indem Sie die Konfigurationsvariable
'advice.detachedHead' auf 'false' setzen.

Library Manager: wmbus-drivers@0.0.0+20250817160729.sha.c856f7d has been installed!
INFO wmbus-drivers@0.0.0+20250817160729.sha.c856f7d has been installed!
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
[... 77 lines cut ...]
* Web  > https://registry.platformio.org/search?q=header:%1B%5Bm%1B%5BKELECHOUSE_CC1101_SRC_DRV.h
*
****************************************************************************************

    3 | #include <ELECHOUSE_CC1101_SRC_DRV.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/wmz-deltamess-s3/src/esphome/components/wmbus/rf_cc1101.cpp.o] Error 1
In file included from src/esphome/components/wmbus/wmbus.cpp:1:
src/esphome/components/wmbus/wmbus.h:17:10: fatal error: WiFiClient.h: No such file or directory

**************************************************************************
* Looking for WiFiClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClient.h"
* Web  > https://registry.platformio.org/search?q=header:%1B%5Bm%1B%5BKWiFiClient.h
*
**************************************************************************

   17 | #include <WiFiClient.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/wmz-deltamess-s3/src/esphome/components/wmbus/wmbus.cpp.o] Error 1
========================= [FAILED] Took 11.21 seconds =========================
`

SzczepanLeon, 2025-08-17:
Because for v3, v4 only arduino framework is supported.

ScullyMulder38, 2025-08-17:
I changed to this type:

ESP32 NodeMCU Module WLAN WiFi Development Board | Dev Kit C V2 mit CP2102 kompatibel mit Arduino

Same error! As soon as I enter this code in the yaml, the error appears:
`wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO5
  clk_pin:  GPIO2
  cs_pin:   GPIO14
  gdo0_pin: GPIO15
  gdo2_pin: GPIO16`

`Compiling .pioenvs/deltamesswmznmcu/src/esphome/components/wmbus/rf_cc1101.cpp.o
In file included from src/esphome/components/wmbus/rf_cc1101.h:9,
                 from src/esphome/components/wmbus/rf_cc1101.cpp:1:
src/esphome/components/wmbus/cc1101_rf_settings.h:3:10: fatal error: ELECHOUSE_CC1101_SRC_DRV.h: No such file or directory

****************************************************************************************
* Looking for ELECHOUSE_CC1101_SRC_DRV.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ELECHOUSE_CC1101_SRC_DRV.h"
* Web  > https://registry.platformio.org/search?q=header:%1B%5Bm%1B%5BKELECHOUSE_CC1101_SRC_DRV.h
*
****************************************************************************************

    3 | #include <ELECHOUSE_CC1101_SRC_DRV.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/deltamesswmznmcu/src/esphome/components/wmbus/rf_cc1101.cpp.o] Error 1
========================= [FAILED] Took 22.13 seconds =========================`

It looks like something is missing?

`src/esphome/components/wmbus/cc1101_rf_settings.h:3:10: fatal error: ELECHOUSE_CC1101_SRC_DRV.h: No such file or directory
`