SzczepanLeon/esphome-components #253: ESP32 + ESPHome 2025.8 (Arduino) + CC1101 + V4: it works (for me ;-))

jacyker, 2025-08-21:
There was a problem reading on 2025.7, so the only sensible solution was to stay on 2025.6.x. Today I tested with 2025.8 and... it works. It reads water (mkradio4). And since ESPHome 2025.7 and 2025.8 offer a lot of cool new features, it's worth checking out, as it's a shame to be stuck with 2025.6.

My config: ESP32 + ESPHome 2025.8 (framework Arduino) + CC1101 @ V4 (mkradio4).

solarssk, 2025-08-21:
can you share your code?

th0m4sek, 2025-08-21:
Not working for
`esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino`
 ```
Compiling .pioenvs/esp32cc1101/lib4c5/Network/NetworkClient.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
Compiling .pioenvs/esp32cc1101/lib4c5/Network/NetworkEvents.cpp.o
*** [.pioenvs/esp32cc1101/lib4c5/Network/NetworkClient.cpp.o] Error 127
sh: 1: riscv32-esp-elf-g++: not found
*** [.pioenvs/esp32cc1101/lib4c5/Network/NetworkEvents.cpp.o] Error 127
```

jacyker, 2025-08-21:
> can you share your code?

`esp32:
  board: nodemcu-32s
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@version_4 # ESP32, Arduino, CC1101
    components: [ wmbus ]

time:
  - platform: sntp
    id: time
    timezone: Europe/Warsaw
    servers:
     - tempus1.gum.gov.pl
     - tempus2.gum.gov.pl

wmbus:
  mosi_pin: GPIO23
  clk_pin: GPIO18
  miso_pin: GPIO19
  gdo2_pin: GPIO27
  gdo0_pin: GPIO14
  cs_pin: GPIO17

  sync_mode: True

sensor:
  - platform: wmbus
    meter_id: 0x12345678
    type: mkradio4
    sensors:
      - name: "CC1101 RSSI"
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"
      - name: "CC1101 Water"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"`

jacyker, 2025-08-21:
> can you share your code?

```
esp32:
  board: nodemcu-32s
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@version_4 # ESP32, Arduino, CC1101
    components: [ wmbus ]
```

jair1111, 2025-08-24:
I think the key part is:

`  sync_mode: True`

Not sure what it does, it simply was the only difference between my config and two posted working ones. I have added this line to my config and data started flowing. YMMV.

Nelson71a, 2025-08-25:
Confirmed.

On version 2025.8 it works for me without making any changes.

jawilczek, 2025-08-26:
great information - it came to life for me too

jacyker, 2025-09-05:
> works for me too, but only if the logger level is not set to verbose

I've also noticed this recently, that logging greater than DEBUG generally causes problems with the entire ESP (when using this add-on/CC1101).