SzczepanLeon/esphome-components #254: V5 + SX1276 + flowiq2200 = Not working

anhorbc, 2025-08-22:
Meter type, media and ID is identified correctly, but no values extracted. If I analyze the telegram on wmbusmeters I get CRC error.

[19:36:36][I][wmbus:044]: Have data from radio (93 bytes) [RSSI: -85dBm, mode: C1]
[19:36:36][D][main:351]: RSSI: -85dBm T: 4f44372c773954233a168d20af4489cd2162d49f384d2e02e5df2e8a3a160901ed669f727592b8ecd4be8e6d2c9f8f933201277fa24ff9cdbfe3aea3854695b6f86421ba6e9f4e491ca74420c9d97080e79c4def5b243115011dfe5786 (93)
[19:36:36][D][wmbusmeters:779]: (meter) flowiq2200-23543977(0) flowiq2200  handling telegram from 23543977.M=KAW.V=3a.T=16
[19:36:36][D][wmbusmeters:267]: (telegram) DLL L=4f C=44 (from meter SND_NR) M=2c37 (KAW) A=23543977 VER=3a TYPE=16 (Cold water meter) (driver flowiq2200) DEV=ESP32-WaterFlow RSSI=-85
[19:36:36][D][wmbusmeters:286]: (telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=af
[19:36:36][D][wmbusmeters:292]:  SN=4489cd21 (AES_CTR session=4 time=1890452) CRC=269d
[19:36:36][I][wmbus:052]: Telegram handled by 1 handlers

[19:38:11][I][wmbus:044]: Have data from radio (64 bytes) [RSSI: -87dBm, mode: C1]
[19:38:11][D][main:351]: RSSI: -87dBm T: 3644372c773954233a168d20b05889cd21c7dc51b4d226ed9998a3d0b1df9fc7b347b411e8a8be8445a97e10a4f27fac407b920a4bf11241cd88e959c1fa822f (64)
[19:38:11][D][wmbusmeters:779]: (meter) flowiq2200-23543977(0) flowiq2200  handling telegram from 23543977.M=KAW.V=3a.T=16
[19:38:11][D][wmbusmeters:267]: (telegram) DLL L=36 C=44 (from meter SND_NR) M=2c37 (KAW) A=23543977 VER=3a TYPE=16 (Cold water meter) (driver flowiq2200) DEV=ESP32-WaterFlow RSSI=-87
[19:38:11][D][wmbusmeters:286]: (telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=b0
[19:38:11][D][wmbusmeters:292]:  SN=5889cd21 (AES_CTR session=8 time=1890453) CRC=f6d9

mortenx, 2025-09-02:
Something still wrong with v5 and kamstrup
The reception of V5 + SX1276 is much better than V4 and CC1101, but am I doing something wrong that I am not getting the right package?
ESPHOME Code:

`

external_components:
  - source: github://SzczepanLeon/esphome-components@main
esp32:
  board: heltec_wifi_lora_32_V2
  flash_size: 8MB
  framework:
    type: esp-idf

# Enable logging
logger:
  id: component_logger
  level: DEBUG
  baud_rate: 115200

spi:
  clk_pin:
    number: GPIO5
    ignore_strapping_warning: true
  mosi_pin: GPIO27
  miso_pin: GPIO19

wmbus_radio:
  radio_type: SX1276
  cs_pin: GPIO18
  reset_pin: GPIO14
  irq_pin: GPIO35
  on_frame:
    - then:
        - logger.log:
            format: "RSSI: %ddBm T: %s (%d)"
            args: [ frame->rssi(), frame->as_hex().c_str(), frame->data().size() ]
    - then:
        - repeat:
            count: 3
            then:
              - output.turn_on: status_led
              - delay: 100ms
              - output.turn_off: status_led
              - delay: 100ms
    - mark_as_handled: True
      then:
        - mqtt.publish:
            topic: wmbus-test/telegram_rtl
            payload: !lambda return frame->as_rtlwmbus();

wmbus_meter:
  - id: water_meter
    meter_id: 0x53589996
    type: Flowiq2200
    key: "06B13E88EE47D4FEAEA56F883A320F5C"
    mode: 
      - T1
      - C1

output:
  - platform: gpio
    id: vext_output
    pin: GPIO21
  - platform: gpio
    id: oled_reset
    pin: GPIO16
    inverted: True
  - platform: gpio
    id: status_led
    pin: GPIO25

sensor:
  - platform: wmbus_meter
    parent_id: water_meter
    field: total
    device_class: water
    name: water_meter_t
    accuracy_decimals: 4
    state_class: total_increasing

`

LOGGER ... V4 esp32+cc1101:
key is : 06B13E88EE47D4FEAEA56F883A320F5C

`[12:25:19][I][wmbus:100]: flowiq2200 [0x53589996] RSSI: -51dBm T: 3444372C969958533C168D20FFD8707B20D51DB6178F02CA4FCF10887D181B41E0BFF3ACC625F356F28CFA84BECB415A28FC87D666 (53) C1 B
[12:25:19][D][meters.cpp:1985]: (meter) created ESPHome flowiq2200 53589996 encrypted
[12:25:19][D][meters.cpp:909]: (meter) ESPHome(0) flowiq2200  handling telegram from 53589996.M=KAW.V=3c.T=16
[12:25:19][D][Telegram.cpp:563]: (telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=ff
[12:25:19][D][Telegram.cpp:570]:  SN=d8707b20 (AES_CTR session=8 time=505613) CRC=8518
[12:25:19][D][Telegram.cpp:2028]: (wmbus) ignoring compressed telegram since format signature hash 0xa9f3 is yet unknown.
[12:25:19][D][Telegram.cpp:2028]:      this is not a problem, since you only need wait for at most 8 telegrams
[12:25:19][D][Telegram.cpp:2028]:      (8*16 seconds) until an full length telegram arrives and then we know
[12:25:19][D][Telegram.cpp:2028]:      the format giving this hash and start decoding the telegrams properly.
[12:25:19][W][wmbus:156]: Can't get requested field 'flow' with unit 'm3h'
[12:25:19][W][wmbus:156]: Can't get requested field 'min_flow_temperature' with unit 'C'
[12:25:19][D][sensor:103]: 'eFlowiq2200 RSSi': Sending state -51.00000 dBm with 0 decimals of accuracy
[12:25:19][W][wmbus:156]: Can't get requested field 'total' with unit 'm³'`

LOGGER ...  V5 + SX1276 + flowiq2200:

`[12:25:19][I][wmbus:044]: Have data from radio (64 bytes) [RSSI: -71dBm, mode: C1]
[12:25:19][D][main:507]: RSSI: -71dBm T: 3644372c969958533c168d20ffd8707b20d51db6178f02ca4fcf10887d181b41e0bff3acc625f356f28cfa84becb415a28fc87d66661c4722e228359c12a1694 (64)
[12:25:19][D][wmbusmeters:779]: (meter) flowiq2200-53589996(0) flowiq2200  handling telegram from 53589996.M=KAW.V=3c.T=16
[12:25:19][D][wmbusmeters:267]: (telegram) DLL L=36 C=44 (from meter SND_NR) M=2c37 (KAW) A=53589996 VER=3c TYPE=16 (Cold water meter) (driver flowiq2200) DEV=wmbus_heltech_v2 RSSI=-71
[12:25:19][D][wmbusmeters:286]: (telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=ff
[12:25:19][D][wmbusmeters:292]:  SN=d8707b20 (AES_CTR session=8 time=505613) CRC=8518`

anhorbc, 2025-09-02:
For V5 frames, the length field is off by +2 bytes. and the frames are too long. Long frames are 15 bytes too long, and short frames are 11 bytes too long.

mortenx, 2025-09-03:
yes same problem,  VERBOSE debug level on the v4, it keeps rebooting and same code and same esp in logger debug level working fine. 
What does this VERBOSE do then?
 v5  VERBOSE  is working fine.
why so?

mortenx, 2025-09-03:
V5

[23:37:09][I][wmbus:100]: flowiq2200 [0x53589996] RSSI: -49dBm T: 4D44372C969958533C168D2026D4F47B20AC50BD50F2695A38C1C0D6F733E83310A4A4AF5445D2FC0C46F12AE5916A2372E9A39D39C74A6D7F9A71CFB1441D89244D1DB5E36AC329CEFE5F940E21 (78) C1 B
[23:37:09][D][meters.cpp:1985]: (meter) created ESPHome flowiq2200 53589996 encrypted
[23:37:09][D][meters.cpp:909]: (meter) ESPHome(0) flowiq2200  handling telegram from 53589996.M=KAW.V=3c.T=16
[23:37:09][D][Telegram.cpp:563]: (telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=26
[23:37:09][D][Telegram.cpp:570]:  SN=d4f47b20 (AES_CTR session=4 time=507725) CRC=b66e
[23:37:09][D][sensor:103]: 'eflowiq2200_volume_flow_m3h': Sending state 0.00000 m3h with 4 decimals of accuracy
[23:37:09][D][sensor:103]: 'eflowiq2200_min_temp_c': Sending state 10.00000 C with 4 decimals of accuracy
[23:37:09][D][sensor:103]: 'eFlowiq2200 RSSi': Sending state -49.00000 dBm with 0 decimals of accuracy
[23:37:10][D][sensor:103]: 'eflowiq2200_total_vesi': Sending state 0.44200 m³ with 4 decimals of accuracy

V4
[23:37:09][V][wmbusmeters:351]: (wmbus) checkWMBUSFrame "4F44372C969958533C168D2026D4F47B20AC50BD50F2695A38C1C0D6F733E83310A4A4AF5445D2FC0C46F12AE5916A2372E9A39D39C74A6D7F9A71CFB1441D89244D1DB5E36AC329CEFE5F940E21C3B91EFAA03C5B24E181BA8A39D315"
[23:37:09][V][wmbusmeters:5094]: (wmbus) received full frame.
[23:37:09][I][wmbus:044]: Have data from radio (93 bytes) [RSSI: -41dBm, mode: C1]
[23:37:09][D][main:507]: RSSI: -41dBm T: 4f44372c969958533c168d2026d4f47b20ac50bd50f2695a38c1c0d6f733e83310a4a4af5445d2fc0c46f12ae5916a2372e9a39d39c74a6d7f9a71cfb1441d89244d1db5e36ac329cefe5f940e21c3b91efaa03c5b24e181ba8a39d315 (93)
[23:37:09][V][wmbusmeters:1066]: (wmbus) parseDLL @0 93
[23:37:09][V][wmbusmeters:1137]: (wmbus) parseELL @10 83
[23:37:09][V][wmbusmeters:554]: (meter) flowiq2200-53589996: for me? 53589996.M=KAW.V=3c.T=16 in 53589996
[23:37:09][V][wmbusmeters:574]: (meter) flowiq2200-53589996: yes for me
[23:37:09][D][wmbusmeters:779]: (meter) flowiq2200-53589996(0) flowiq2200  handling telegram from 53589996.M=KAW.V=3c.T=16
[23:37:09][V][wmbusmeters:784]: (meter) flowiq2200-53589996 53589996.M=KAW.V=3c.T=16 "4F44372C969958533C168D2026D4F47B20AC50BD50F2695A38C1C0D6F733E83310A4A4AF5445D2FC0C46F12AE5916A2372E9A39D39C74A6D7F9A71CFB1441D89244D1DB5E36AC329CEFE5F940E21C3B91EFAA03C5B24E181BA8A39D315"
[23:37:09][V][wmbusmeters:1066]: (wmbus) parseDLL @0 93
[23:37:09][D][wmbusmeters:267]: (telegram) DLL L=4f C=44 (from meter SND_NR) M=2c37 (KAW) A=53589996 VER=3c TYPE=16 (Cold water meter) (driver flowiq2200) DEV=wmbus_heltech_v2 RSSI=-41
[23:37:09][V][wmbusmeters:1137]: (wmbus) parseELL @10 83
[23:37:09][V][wmbusmeters:351]: (ELL) decrypting "AC50BD50F2695A38C1C0D6F733E83310A4A4AF5445D2FC0C46F12AE5916A2372E9A39D39C74A6D7F9A71CFB1441D89244D1DB5E36AC329CEFE5F940E21C3B91EFAA03C5B24E181BA8A39D315"
[23:37:09][V][wmbusmeters:062]: (ELL) IV 372C969958533C1620D4F47B20000000
[23:37:09][V][wmbusmeters:081]: (ELL) block 0 block_size 16 offset 0
[23:37:10][V][wmbusmeters:351]: (ELL) decrypted "B66E7804FF23013A00000413BA010000"
[23:37:10][V][wmbusmeters:351]: (ELL) decrypted "00A1015B0A91015B0EA1016712AC207C"
[23:37:10][V][wmbusmeters:081]: (ELL) block 4 block_size 12 offset 64
[23:37:10][V][wmbusmeters:351]: (ELL) decrypted "0F8E7AF166355B461E6E12BF"
[23:37:10][V][wmbusmeters:351]: (ELL) decrypted "B66E7804FF23013A00000413BA01000044139A010000426C2139023B000092013B1000A2013B000006FF1B05500005C000A1015B0A91015B0EA1016712AC207C0F8E7AF166355B461E6E12BF"
[23:37:10][D][wmbusmeters:286]: (telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=26
[23:37:10][D][wmbusmeters:292]:  SN=d4f47b20 (AES_CTR session=4 time=507725) CRC=b66e
[23:37:10][V][mqtt:513]: Publish(topic='wmbus-test/telegram_rtl' payload='C1;1;1;2025-09-03 20:37:08.00Z;-41;;;0x4f44372c969958533c168d2026d4f47b20ac50bd50f2695a38c1c0d6f733e83310a4a4af5445d2fc0c46f12ae5916a2372e9a39d39c74a6d7f9a71cfb1441d89244d1db5e36ac329cefe5f940e21c3b91efaa03c5b24e181ba8a39d315
[23:37:10]' retain=0 qos=0)

SzczepanLeon, 2025-09-04:
Please try with:
```
external_components:
  - source: github://SzczepanLeon/esphome-components@len_fix
```

DEBUG level should be enough

SzczepanLeon, 2025-09-04:
@mortenx please add couple more lines above "Have data from radio"

from line:
```
[11:06:30][D][packet:092]: expected_size: ab  size: cd
```

anhorbc, 2025-09-04:
Works for me. Did a clean build and it worked right away. 👌

mortenx, 2025-09-05:
> Please include YAML

`
external_components:
  - source: github://SzczepanLeon/esphome-components@main
esp32:
  board: heltec_wifi_lora_32_V2
  flash_size: 8MB
  framework:
    type: esp-idf

# Enable logging
logger:
  id: component_logger
  level: DEBUG
  baud_rate: 115200

time:
  - platform: homeassistant
captive_portal:
    
spi:
  clk_pin:
    number: GPIO5
    ignore_strapping_warning: true
  mosi_pin: GPIO27
  miso_pin: GPIO19

wmbus_radio:
  radio_type: SX1276
  cs_pin: GPIO18
  reset_pin: GPIO14
  irq_pin: GPIO35

  on_frame:
    - then:
        - logger.log:
            format: "RSSI: %ddBm T: %s (%d)"
            args: [ frame->rssi(), frame->as_hex().c_str(), frame->data().size() ]
    - then:
        - repeat:
            count: 3
            then:
              - output.turn_on: status_led
              - delay: 100ms
              - output.turn_off: status_led
              - delay: 100ms
    - mark_as_handled: True
      then:
        - mqtt.publish:
            topic: wmbus-test/telegram_rtl
            payload: !lambda return frame->as_rtlwmbus();

wmbus_meter:
  - id: water_meter
    meter_id: 0x53589996
    type: Flowiq2200
    key: "06B13E88EE47D4FEAEA56F883A320F5C"
    mode: 
      - T1
      - C1

  - id: water_meter2
    meter_id: 0x76395254
    type: multical21
    key: "2B5A564C47E7635F4726F07377F9EBC0"
    mode: 
      - T1
      - C1
output:
  - platform: gpio
    id: vext_output
    pin: GPIO21
  - platform: gpio
    id: oled_reset
    pin: GPIO16
    inverted: True
  - platform: gpio
    id: status_led
    pin: GPIO25

sensor:
  - platform: wmbus_meter
    parent_id: water_meter
    field: "total"
    device_class: water
    name: water_meter_f
    id: water_meter_f    
    accuracy_decimals: 4
    state_class: total_increasing
    icon: "mdi:water"
    unit_of_measurement: "m³"
  
  - platform: wmbus_meter
    parent_id: water_meter  
    name: "flowiq_RSSif_"
    field: "rssi"
    accuracy_decimals: 0
    unit_of_measurement: "dBm"
    device_class: "signal_strength"
    state_class: "measurement"
    entity_category: "diagnostic"

  - platform: wmbus_meter
    parent_id: water_meter2
    field: "total"
    device_class: water
    name: water_meter_f2
    id: water_meter_f2    
    accuracy_decimals: 4
    state_class: total_increasing
    icon: "mdi:water"
    unit_of_measurement: "m³"
 
  - platform: wmbus_meter
    parent_id: water_meter2  
    name: "rssmultical21_RSSif_"
    field: "rssi"
    accuracy_decimals: 0
    unit_of_measurement: "dBm"
    device_class: "signal_strength"
    state_class: "measurement"
    entity_category: "diagnostic"

`

mortenx, 2025-09-05:
Yes now its working, field was wrong

Can you also say how to get meter rssi to sensor working?
How should it be formatted v5, it is not in wmbusmeters

 In V4 was like this:

      - name: "eFlowiq2200 RSSi"
        id: eflowiq2200_rss      
        field: "rssi"
        accuracy_decimals: 0
        unit_of_measurement: "dBm"
        device_class: "signal_strength"
        state_class: "measurement"
        entity_category: "diagnostic"

SzczepanLeon, 2025-09-05:
```
sensor:
  - platform: wmbus_meter
    id: water_meter_rssi
    field: rssi_dbm
    parent_id: my_meter
    name: RSSI
    device_class: signal_strength
    state_class: measurement
```

mortenx, 2025-09-08:
Now its working, flowiq2200 and multical21 :)

sensor:
  - platform: wmbus_meter
    parent_id: water_meter
    field: total_m3
    device_class: water
    name: "flowiq_total_757"
    id: flowiq_total_757    
    accuracy_decimals: 4
    state_class: total_increasing
    icon: "mdi:water"
    unit_of_measurement: "m³"
  
  - platform: wmbus_meter
    parent_id: water_meter  
    name: "flowiq_RSSif_757"
    field: rssi_dbm
    accuracy_decimals: 0
    unit_of_measurement: "dBm"
    device_class: "signal_strength"
    state_class: "measurement"
    entity_category: "diagnostic"

  - platform: wmbus_meter
    parent_id: water_meter  
    name: "flowiq_flow_757"
    field: flow_m3h
    accuracy_decimals: 3
    unit_of_measurement: "l/h"
    filters:
      - multiply: 1000

  - platform: wmbus_meter
    parent_id: water_meter2
    field: "total_m3"
    device_class: water
    name: multical21_total
    id: multical21_total    
    accuracy_decimals: 4
    state_class: total_increasing
    icon: "mdi:water"
    unit_of_measurement: "m³"
 
  - platform: wmbus_meter
    parent_id: water_meter2  
    name: "rssmultical21_RSSif_"
    field: rssi_dbm
    accuracy_decimals: 0
    unit_of_measurement: "dBm"
    device_class: "signal_strength"
    state_class: "measurement"
    entity_category: "diagnostic"

text_sensor:
  - platform: wmbus_meter
    parent_id: water_meter
    field: status
    name: flowiq_status