SzczepanLeon/esphome-components #257: esp32 +V4 + cc1101+ flowiq2200 not decoding flow

mortenx, 2025-08-26:
Total is ok but can ´t decode flow.

 Is field ok?

esphome :

sensor:
  - platform: wmbus
    meter_id: 0x53589996
    type: Flowiq2200
    key: "06B13E88EE47D4FEAEA56F883A320F5C"
    sensors:
      - name: "flowiq2200_total_vesi"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
      - name: "flowiq2200_volume_flow_lh"
        field: "flow"
        accuracy_decimals: 4
        unit_of_measurement: "l/h"
        device_class: "water"
        state_class: "measurement"
        icon: "mdi:water"

Log:

`[17:40:08][I][wmbus:094]: Using selected driver flowiq2200 (detected driver was flowiq2200)
[17:40:08][I][wmbus:100]: flowiq2200 [0x53589996] RSSI: -45dBm T: 3444372C969958533C168D2027880E79207482C72A3E6A75AA1CBDFEDBE5304C6C8B488CB062E5DF5932E733432EFFE324423AA404 (53) C1 B
[17:40:08][D][meters.cpp:1985]: (meter) created ESPHome flowiq2200 53589996 encrypted
[17:40:08][D][meters.cpp:909]: (meter) ESPHome(0) flowiq2200  handling telegram from 53589996.M=KAW.V=3c.T=16
[17:40:08][D][Telegram.cpp:563]: (telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=27
[17:40:08][D][Telegram.cpp:570]:  SN=880e7920 (AES_CTR session=8 time=495848) CRC=2aca
[17:40:08][D][sensor:103]: 'flowiq2200_total_vesi': Sending state 0.35400 m³ with 3 decimals of accuracy
[17:40:08][W][wmbus:156]: Can't get requested field 'volume_flow_lh' with unit 'l/h'`

SzczepanLeon, 2025-08-26:
https://wmbusmeters.org/analyze/4D44372C969958533C168D2026740E7920DAFD7627919FDF066783976DEA0BF8CC5E19A8970003BDFF37852D4555DEA46213E2E2ACD2408B6628B255BAEF8B3488B421159A1AC37D8420A9CEA395:auto:06B13E88EE47D4FEAEA56F883A320F5C

field is OK, but unit is wrong

mortenx, 2025-08-26:
Is it possible to get water temperature also?

mortenx, 2025-08-31:
working :)

sensor:
  - platform: wmbus
    meter_id: 0x53589996
    type: Flowiq2200
    key: "06B13E88EE47D4FEAEA56F883A320F5C"
    sensors:
      - name: "flowiq2200_total_vesi"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class: "total_increasing"
        icon: "mdi:water"
      - name: "flowiq2200_volume_flow_lh"
        field: "flow"
        accuracy_decimals: 4
        unit_of_measurement: "m3h"
        device_class: "water"
        state_class: "measurement"
        icon: "mdi:water"
      - name: "flowiq2200_min_temp_c"
        field: "min_flow_temperature"
        accuracy_decimals: 4
        unit_of_measurement: "C"
        device_class: "water"
        state_class: "measurement"
        icon: "mdi:water"

closed 2025-09-02 by SzczepanLeon